jerrod-lankford/GarageQTPi #2: no option for NC switches

Spartan-II-117, 2017-01-11:
Can a configuration option be created to use NC magnetic switches? (reverse open and closed?)

Jerrkawz, 2017-01-11:
Was considering this initially. I can add an optional config parameter to the door. Thoughts on a name for the parameter? 

Was thinking something like:

sensor_mode: normally_closed

or

invert_sensor: true

Thoughts?

Spartan-II-117, 2017-01-11:
I think sensor_mode with NO or NC as options makes the most sense, with the option existing in the config file either way.

Spartan-II-117, 2017-04-23:
Any chance of getting this implemented?

Jerrkawz, 2017-04-23:
Yeah sorry I got busy and forgot about it until the other issue today. I'll knock it out tomorrow

Spartan-II-117, 2017-04-23:
Awesome, thanks! If I actually knew what I was doing with this stuff I'd have just submitted a PR, but I don't.

closed 2017-04-23 by Jerrkawz