jerrod-lankford/GarageQTPi #3: Issue with re-connects

sabooker, 2017-04-22:
Something I found is that if your mqtt service(mosquitto) what have you re-connects or restarts, this will cause your garage door to open if closed/close if opened. This is a pretty major issue. Right now I have an automation in homeassistant that will auto-close the doors if they are open for a given amount of time, but it is still pretty scary that if you have a power outage at night my garage doors will open when mosquitto comes back online.

I think it could be fixed by adding a variable that holds the current mqtt value and checks against new values received. If they match then it does not trigger??

Jerrkawz, 2017-04-23:
Do you think its because the state moves from closed to unknown? It should be an unknown state if the mqtt connection is lost and it has to reconnect. I think I can use the retention flag when i publish a message to mqtt and it will save the last message, so that if your connection is reset it should read the last message and see that the state is whatever it was. I've been meaning to do this anyway. I'll post back here when ive committed it and you can try it?

Jerrkawz, 2017-04-23:
Alright I added the retain flag to published messages. Now when i restart home assistant my garage state is read correctly from the last previous message. This is just nice by itself but can you also let me know if it fixes your problem?

Jerrkawz, 2017-04-23:
I am also not opposed to adding what you suggested and only opening the door if the state is currently reported as closed from the pin, but I might add it as an additional config option.

strict: true

or something like that. Thoughts?

Spartan-II-117, 2017-04-23:
Setting the retain flag fixes the issue of HASS restarting, but I don't believe it handles if mosquito restarts.

Jerrkawz, 2017-04-23:
True but if mosquito restarts then it shouldn't affect the current state in HASS

sabooker, 2017-04-23:
It is more a problem because the code for "open" and the code for "closed" is the same, so when you sent a mqtt message out saying "closed" and the door is closed it will still open it. The code just brings the gpio low, pauses, then brings it high no matter what. 

When your mqtt broker restarts it drops the connections, so when GarageQTPi reconnects one of the first thing it does is subscribe to the topics, this causes it to recieve a message "closed" for the door. This then initiates the door opening because of problem above.

I don't really think that you can solve the issue by modifying the gpio code, you would have to put something in to basically ignore the initial message you get after re-subscribing.

Jerrkawz, 2017-04-23:
This seems potentially more like an issue with mosquitto? I haven't seen this behavior in cloudmqtt. This behavior of receiving a message on subscribe should only happen with the retained messages (which are only the state messages, not the command messages).

I can investigate further but I may need to set up a local mosquitto instance.

sabooker, 2017-04-23:
Is there a specific change I can make in mosquitto or HA that you guys are aware of that can change this?

Jerrkawz, 2017-04-23:
You would have to read the documentation or ask mosquitto if they are retaining messages that shouldn't be retained.

Are you sure that whats happening is mosquito is restarting and its resending a command? Or could it have been that home assistant was restarting and not reading the state properly? Until I made my change yesterday if home assistant were to restart, the state of the garage door entity would have been UNKNOWN, until it was opened or closed again. So is it possible that your automation that automatically closes the door was seeing closed != UNKNOWN and triggering the close command? If this is the case then it should be resolved by my change yesterday.

Spartan-II-117, 2017-05-09:
Has this been resolved @sabooker?

onlize, 2017-06-19:
sabooker, can you please tell me how did you configure automation in homeassistant that will auto-close the door if it is open for a given amount of time?

Thank you.

onlize, 2017-07-11:
I have exactly the same issue. If I restart HA or if I have a power failure for GarageQTPi, but not for HA, the door closes if it was open and it will open if it was closed.

What can I do to fix?

Jerrkawz, 2017-07-11:
Are you also using mosquitto?

onlize, 2017-07-12:
Yes, I use mosquitto and it is on the same RPi as Home Assistant.

onlize, 2017-07-12:
I do not think that issue is with mosquitto. This is an example, All I did is to stopped and started the script. It can see that the door is closed and it is trying to close it, but the door opens. You can see the CLOSE comes as soon as I started this script. After that, I commanded close the door and you see the second CLOSE. my guess, this script does not understand the difference between close and open the door. It just knows that there is should be a change:

pi@GarageQTPi:~/GarageQTPi $ python main.py
Welcome to GarageBerryPi!
/home/pi/GarageQTPi/lib/garage.py:29: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.relay_pin, GPIO.OUT)
Connected with result code: Connection Accepted.
Listening for commands on home-assistant/cover/set
Executing command CLOSE for door Garage Door
State change triggered: home-assistant/cover -> closed
State change triggered: home-assistant/cover -> open
Executing command CLOSE for door Garage Door
State change triggered: home-assistant/cover -> open
State change triggered: home-assistant/cover -> closed

Jerrkawz, 2017-07-12:
But see thats not the way mqtt should work.

Yes you are right the script has no idea the difference between the open and close command. It just flips the pin on and off quickly to simulate a button press. BUT the thing is when you subscribe to a thread on mqtt it SHOULD NOT get a new message right away UNLESS the message sent had a retain flag. So something in the mosquitto service is seeing the script connecting and going "Oh hey here was the last message "CLOSE" and the script goes "OH ok fuck it ill just execute this code"

Thats why my above message was about implementing a strict mode where i actually read the pin state every time a message is sent and then decide if i want to actually execute the command or not. Super easy to implement but I haven't had time. Maybe i can get around to it tonight

Jerrkawz, 2017-07-12:
To clarify theres two solutions:

1. Figure out why mosquitto is behaving this way (I bet its some flag that i need to set on connect)
2. Implement a strict mode where executing a close or open command would check the state of the magnets before executing the code.

onlize, 2017-07-12:
Here is when I have during testing. Every time, when I run main.py script, relay switches position. I guess that this is the moment when it opens/closes the door. Also, something else:

If I connect Garage Door to NC position of the relay.
 1. The door opens if closed and closes when opened.
 2. Because this is NC position, I cannot close garage door using the button on the wall if the program is not running.

If I connect Garage Door to NO position of the relay.
 1. The door does not change position
 2. Because this is now NC position, I cannot close garage door using the button on the wall if the program is running.

onlize, 2017-07-12:
Any way to change the program, so it would not change the relay's position every time it starts? I believe it would solve all problems.

onlize, 2017-07-13:
It release the relay every time I shutdown RPi. And it changes the position every time it starts.

Because of this, every time I reboot my RPi, it changes position. after that, when everything is coming back up, it changes position again.

Jerrkawz, 2017-07-13:
Well was basing my assumption off of your log lines. Notice the script connects then the close command happens

Connected with result code: Connection Accepted.
Listening for commands on home-assistant/cover/set
Executing command CLOSE for door Garage Door
State change triggered: home-assistant/cover -> closed
State change triggered: home-assistant/cover -> open
Executing command CLOSE for door Garage Door
State change triggered: home-assistant/cover -> open
State change triggered: home-assistant/cover -> closed

So thats whats making me think that mosquitto is sending us a CLOSE message when we connect.

onlize, 2017-07-13:
I think it is just changing position. It sends CLOSE if the door is opened. If the door is closed, I always have this:

pi@GarageQTPi:~/GarageQTPi $ python main.py
Welcome to GarageBerryPi!
Connected with result code: Connection Accepted.
Listening for commands on home-assistant/cover/set
Executing command OPEN for door Garage Door
State change triggered: home-assistant/cover -> closed
State change triggered: home-assistant/cover -> open

onlize, 2017-07-13:
I think, because relay changes position during the start, it thinks that the button was pressed and does what it should do when the button is pressed.

sabooker, 2017-07-13:
Sorry for very late reply. I did not get this issue resolved. I am actually moving towards building a controller based off an esp8266/mqtt at this time. Smaller package, so I can re-use the raspberry pi for something a bit more complex.

sabooker, 2017-07-13:
Also here is my yaml code for the auto close after a specific time.

- id: Right_Garage_Door_Auto_Close
  alias: 'Right Garage Door Open'
  trigger:
    platform: state
    entity_id: cover.right_garage_door
    state: 'open'
    for:
      minutes: 30
  condition:
    condition: and
    conditions:
      - condition: state
        entity_id: alert.right_garage_door
        state: 'on'
  action:
    service: cover.close_cover
    entity_id: cover.right_garage_door

onlize, 2017-07-13:
@sabooker

Interesting. I do not know much about esp8266, but I just two days ago I bought one to see what can it do and to learn it. What software will you use?

Jerrkawz, 2017-07-14:
@sabooker @onlize Can one of you two (or both of you) post the mqtt cover configuration from your home assistant configuration.yaml?

onlize, 2017-07-14:
Here is my cover configuration:

cover:
  - platform: mqtt
    state_topic: "home-assistant/cover"
    command_topic: "home-assistant/cover/set"
    name: "Garage Door"
    qos: 0
    retain: true

For some this post shows fat dot instead of the dash.

onlize, 2017-07-14:
I would really like to fix this if possible because everything is already in place.

Jerrkawz, 2017-07-14:
As I thought :P Please remove retain: true.

But your topic has a retained message already and I'm not sure if home assistant clears them or not (i dont). So if deleting retain=True doesn't fix it then:

Try running this command 'mosquitto_pub -t home-assistant/cover/set -m "" -r'
and try it again.

If that doesn't work you can always change your command topic to a new one.

onlize, 2017-07-14:
I removed retain=True but it did not help.

Sorry, do not understand the rest. Where should I run that mosquitto command? Should it be on Home Assistant? Should I reboot and test after that?

Also, if I change the command topic in HA's configuration file, will GarageQTPi work? It expects that one I have.

Jerrkawz, 2017-07-14:
You run that command on the device that is running mosquitto. Its a utility that should have been installed along with mosquitto. If you change the command topic in HA you also need to change it in the garage qt pi config file.

Jerrkawz, 2017-07-14:
You can also restart your mosquitto server. That should clear out any retained messages

onlize, 2017-07-14:
After I removed retain=True, I tested and the test failed.

After that, I restarted my Home Assistant and tested again. The test failed. 

I guess there is no reason to run the mosquitto_pub command or change topic, am I correct?

Jerrkawz, 2017-07-14:
The test will fail until you remove the retained message on your mosquitto server. Try restarting it or running the pub command. Either should work

onlize, 2017-07-14:
Do you still want me to run it even if I restarted HA?

I tried to run it, but I have the following errors:
pi@Home-Assistant:~ $ mosquitto_pub -t home-assistant/cover/set -m "" -r
Connection Refused: not authorised.
Error: The connection was refused.
pi@Home-Assistant:~ $ sudo mosquitto_pub -t home-assistant/cover/set -m "" -r
Connection Refused: not authorised.
Error: The connection was refused.
pi@Home-Assistant:~ $

Jerrkawz, 2017-07-14:
Ah yeah you will need to authenticate. Apologies.

sudo mosquitto_pub -t home-assistant/cover/set -m "" -r -u {your_username} -P {your_password}

Heres the documentation to the command in case you need other flags:

https://mosquitto.org/man/mosquitto_pub-1.html

onlize, 2017-07-14:
I run that command, tested and the test fails.

When I test it, I reboot GarageQTPi. When it goes down, the door changes it status. When it comes back on, it changes it again.

onlize, 2017-07-14:
Sorry, it changes position when it goes down. It does not change position when it the program starts

Jerrkawz, 2017-07-14:
So it looks like we fixed the reconnect issue at least. Now It sounds like the pin flips whenever the program is exited. I would have to investigate this one separately but I'm done for tonight. Out of curiosity which pin are you using for the relay? GPIO 23?

onlize, 2017-07-14:
Yes, pin 23

Jerrkawz, 2017-07-14:
This thread might be helpful if you wanted to take a stab at debugging it: https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=35321

onlize, 2017-07-14:
I thought of that, but, if your program does not start, the GPIO pin is low. It looks to me that when I run main.py, the GPIO pin switches to HIGH and this causes the issue.

I do not know Python, so not sure if it is easy to leave it LOW and, when button is pressed, switch it to HIGH for short period of time.

onlize, 2017-07-14:
It would also solve the problem with button for the Garage door. It would work always and it would not matter if program is running or not.

onlize, 2017-07-14:
I am able to make the door work correctly and do not change the relay when it starts, but this breaks the status of the door. Cannot figure out why.

Jerrkawz, 2017-07-14:
What did you change to make the door work properly

onlize, 2017-07-14:
I fixed it. I made the following changes in garage.py file. I think you have to change it in your code on this website:

###        GPIO.output(self.relay_pin, True)
        GPIO.output(self.relay_pin, False)
.
.
.
.
###        GPIO.output(self.relay_pin, False)
        GPIO.output(self.relay_pin, True)
        time.sleep(SHORT_WAIT)
###        GPIO.output(self.relay_pin, True)
        GPIO.output(self.relay_pin, False)

onlize, 2017-07-14:
I fixed it. I made the following changes in garage.py file. I think you have to change it in your code on this website:
GPIO.output(self.relay_pin, False)
.
.
.
GPIO.output(self.relay_pin, True)
time.sleep(SHORT_WAIT)
GPIO.output(self.relay_pin, False)

Jerrkawz, 2017-07-14:
@onlize Sounds like your relay must be wired differently than mine. Normally opened vs normally closed. Glad you got it working.

onlize, 2017-07-14:
Forgot to say. The door should be connected to normally opened.

The way your program is now: If you connect garage door when relay is normally closed, it means that when your program is not running, you cannot use the garage door button.

Jerrkawz, 2017-07-14:
If the relay is wired in parallel with the button it should work either way. Source: mine works fine with or without the program running. Just checked

onlize, 2017-07-15:
Did you try to shutdown your RPi and try it before the program starts?

You are correct that in parallel it should work either way. Let's think about this.

The way I have it connected and it does not switch relay if the program starts:
I have it connected to NO. It does not switch relay position when the program starts.
In this case I have both buttons (relay and garage door button) opened and does not matter which button you press, it will work.
It will do it in case if program started and if program did not start.

The way you have it working:
If you connect garage door in NO position:
Before you start your program, garage door button works because both contacts (relay and garage door button) are opened. After you start your program, relay closes and garage door button will not work. It is the same as you pressed a button and hold it and, at the same time, trying to press another button.
 If you connect your door in NC position:
Your garage door will not work because one of the buttons already pressed. When you start your program, it releases relay and both buttons are opened and you can use either one of them.

Jerrkawz, 2017-07-15:
My button works fine program on or program off, even with the pi off.

If they are in parallel the closed circuit has no affect on the other circuit. A garage door opener is fairly simple, it just applies voltage to the circuit and triggers the switch. When you press the button it completes a circuit and applies voltage. When you trigger the relay it completes the circuit (for .2 seconds) just as if you pressed a button. Since the button and the relay circuit are both connected directly to the garage door terminals applying voltage on either circuit (regardless of the state of the other circuit) will trigger the door.

What could cause the button to stop working is if the relay is closed, thus constantly supplying voltage to the garage door then it would be just like you were holding your door button down. So hitting a second button wouldn't do any good.

onlize, 2017-07-15:
If one of the buttons is closed, and another one is in parallel, there is no way the second button can have any influence when it is pressed.

Jerrkawz, 2017-07-15:
Yes we are saying the same thing. If you want to keep talking lets take it to email. jllankfo@ncsu.edu. I'm going to repost the solution for the retain problem in case anyone else has it and close this out.

Jerrkawz, 2017-07-15:
For anyone who has this issue make sure the home assistant configuration for the cover does not have retain: true. If it does remove it and following these steps:

Try running this command 'mosquitto_pub -t home-assistant/cover/set -m "" -r -u username -P password'
and try it again. (if you aren't using mosquitto locally you will have to find another way to publish an empty message to your command thread with the retain flag to clear out any retained messages)

If that doesn't work you can always change your command topic to a new one.

closed 2017-07-15 by Jerrkawz