jerrod-lankford/GarageQTPi #8: Cannot install

onlize, 2017-06-18:
Hello,

I am trying to install it and the problems.

    I installed Raspbian
    sudo apt-get update
    sudo apt-get upgrade
    sudo apt-get install python3-dev python3-venv
    python3 -m venv env
    env/bin/python -m pip install --upgrade pip setuptools
    git clone https://github.com/Jerrkawz/GarageQTPi.git

Next, I am trying to run: pip install -r requirements.txt, I receive the following error:
Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'
Storing debug log for failure in /root/.pip/pip.log

After that, I figured out that I have to go to GarageQTPi directory first and run it from that directly. This is the output and error:

pi@Garage-Door:~$ pip install -r GarageQTPi/requirements.txt
Downloading/unpacking paho-mqtt==1.2 (from -r GarageQTPi/requirements.txt (line 1))
  Downloading paho-mqtt-1.2.tar.gz (49kB): 49kB downloaded
  Running setup.py (path:/tmp/pip-build-xftl86/paho-mqtt/setup.py) egg_info for package paho-mqtt

Downloading/unpacking PyYAML==3.12 (from -r GarageQTPi/requirements.txt (line 2))
  Downloading PyYAML-3.12.tar.gz (253kB): 253kB downloaded
  Running setup.py (path:/tmp/pip-build-xftl86/PyYAML/setup.py) egg_info for package PyYAML

Installing collected packages: paho-mqtt, PyYAML
  Running setup.py install for paho-mqtt
    error: could not create '/usr/local/lib/python2.7/dist-packages/paho': Permission denied
    Complete output from command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-xftl86/paho-mqtt/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-tJzjZ4-record/install-record.txt --single-version-externally-managed --compile:
    running install

running build

running build_py

creating build

creating build/lib.linux-armv7l-2.7

creating build/lib.linux-armv7l-2.7/paho

copying src/paho/__init__.py -> build/lib.linux-armv7l-2.7/paho

creating build/lib.linux-armv7l-2.7/paho/mqtt

copying src/paho/mqtt/client.py -> build/lib.linux-armv7l-2.7/paho/mqtt

copying src/paho/mqtt/subscribe.py -> build/lib.linux-armv7l-2.7/paho/mqtt

copying src/paho/mqtt/__init__.py -> build/lib.linux-armv7l-2.7/paho/mqtt

copying src/paho/mqtt/publish.py -> build/lib.linux-armv7l-2.7/paho/mqtt

running install_lib

creating /usr/local/lib/python2.7/dist-packages/paho

error: could not create '/usr/local/lib/python2.7/dist-packages/paho': Permission denied

----------------------------------------
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-xftl86/paho-mqtt/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-tJzjZ4-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-xftl86/paho-mqtt
Storing debug log for failure in /home/pi/.pip/pip.log

Can you help?
Thank you.

Spartan-II-117, 2017-06-18:
You either need to run it within the venv you installed earlier, or sun pip as root "sudo pip..."

onlize, 2017-06-18:
sudo worked for me, thank you.

closed 2017-06-18 by onlize