jerrod-lankford/GarageQTPi #9: Some error when I try to sudo python main.py

onlize, 2017-06-18:
Looks to me that it does not work. Can you please take a look at this and let me know?

pi@Garage-Door:~/GarageQTPi $ sudo python main.py
Welcome to GarageBerryPi!
Traceback (most recent call last):
  File "main.py", line 37, in <module>
    CONFIG = yaml.load(ymlfile)
  File "/usr/local/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/local/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 64, in compose_node
    if self.check_event(AliasEvent):
  File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 449, in parse_block_mapping_value
    if not self.check_token(KeyToken, ValueToken, BlockEndToken):
  File "/usr/local/lib/python2.7/dist-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/local/lib/python2.7/dist-packages/yaml/scanner.py", line 224, in fetch_more_tokens
    return self.fetch_alias()
  File "/usr/local/lib/python2.7/dist-packages/yaml/scanner.py", line 607, in fetch_alias
    self.tokens.append(self.scan_anchor(AliasToken))
  File "/usr/local/lib/python2.7/dist-packages/yaml/scanner.py", line 926, in scan_anchor
    % ch.encode('utf-8'), self.get_mark())
yaml.scanner.ScannerError: while scanning an alias
  in "config.yaml", line 3, column 11
expected alphabetic or numeric character, but found '\n'
  in "config.yaml", line 3, column 12
pi@Garage-Door:~/GarageQTPi $

I do not have the hardware connected yet. Could this be an issue? Everything is in mail, should be here shortly, but I want to be sure that I am ready. How can I check if it works correctly?
Thank you.

Spartan-II-117, 2017-06-18:
Looks like you have an error in your config file on line 3

Jerrkawz, 2017-06-18:
Can you post your config.yaml?

onlize, 2017-06-18:
I thought I had to use config.yaml exactly as it is in the instruction, but I guess not. The error on line 3 was related to the port in Home Assistant that I use. As soon as I changed it, it says the error is on line number 4. How can I find what is the user name and password for the MQTT Cover that I configured in Home Assistant?

This is my config.yaml file:

mqtt:
    host: 192.168.51.12
    port: 1883
    user: *
    password: *
doors:
    -
        id: 'Garage Door'
        relay: 23
        state: 17
        state_mode: normally_closed #defaults to normally open, uncomment to sw$
        state_topic: "home-assistant/cover"
        command_topic: "home-assistant/cover/set"

Thank you.

onlize, 2017-06-18:
I found it, thank you for pointing me right direction.

closed 2017-06-18 by onlize