jerrod-lankford/GarageQTPi #10: The door would not open/close and Home Assistant cannot see the status of the door.

onlize, 2017-07-04:
I did not connect my garage door yet. I am listening to the relay and checking it with tester. Also moving sensors to change status open/close the garage door.

I am not sure if I have to open multiple tickets, or can I do it in one. I have a few issues.

1. The status of the door is not reported to the Home Assistant.
2. Relay that opens the garage door does not change.

This is the status of the door on the Raspberry Pi with GarageQTPi installed:
[image]

I see this when I try to open/close door from Home Assistant. I also move sensor to change the status:
[image]

I see the following error in Home Assistant every time I change status of the door (Move sensors):
`2017-07-03 23:46:37 WARNING (MainThread) [homeassistant.components.cover.mqtt] Payload is not True, False, or integer (0-100): 2017-07-03 23:47:03 WARNING (MainThread) [homeassistant.components.cover.mqtt] Payload is not True, False, or integer (0-100): 2017-07-03 23:47:13 WARNING (MainThread) [homeassistant.components.cover.mqtt] Payload is not True, False, or integer (0-100): 2017-07-03 23:47:17 WARNING (MainThread) [homeassistant.components.cover.mqtt] Payload is not True, False, or integer (0-100): 2017-07-03 23:47:28 WARNING (MainThread) [homeassistant.components.cover.mqtt] Payload is not True, False, or integer (0-100): `

This is my config.yaml file:
[image]

This is the entry for this program in configuration.yaml file in Home Assistant:
[image]

Based on the error in Home Assistant, it seems that it wants to get True, False, or integer and not close, open, or state. Not sure about this.

Can you please help?

Thank you.

onlize, 2017-07-04:
I changed configuration file on Home Assistant to:
cover:
  - platform: mqtt
    state_topic: "home-assistant/cover"
    command_topic: "home-assistant/cover/set"
    name: "Garage Door"
    qos: 0
    retain: true

Now it detects the status of the door, but relay would not change. As soon as I run:
sudo python main.py
It closes "Normally Opened" part of relay and keeps it like this until I reboot my RPi.

Any ideas?
Thank you.

Jerrkawz, 2017-07-05:
Are you sure the comments are getting published to the topic?

onlize, 2017-07-05:
I believe everything is published the way it should. When I started, I had some additional issues, but I resolved them, so I removed some text.

Where can I get those logs?

Thank you.

Jerrkawz, 2017-07-06:
Well you can just post a screen cap of the terminal for garage qt pi. The home assistant log will depend on how its installed. Min is in /var/opt/home-assistant/home-assistant.log

onlize, 2017-07-06:
I cannot find home-assistant.log file, but this is the terminal screenshot from garage qt pi:
[image]

Thank you.

Jerrkawz, 2017-07-06:
I see a warning at the top about the gpio pin being in use. Can you try restarting the pi and rerunning the program?

onlize, 2017-07-06:
I restarted and run it multiple times. I always have this warning and the same behavior..

As soon as I run the program, this pin start giving voltage. It does not happen until I start the program. The only way for me to stop it, is to reboot the RPi. Even if I try Ctrl+C to stop the program, it would not release the pin. I have to restart.

Thank you.

onlize, 2017-07-06:
It seems that the pin actually disconnect the voltage for short period of time if relay is not connected. If relay is connected, the voltage does not change. Does it mean that I have a bad relay? What do you think?

This is a single channel relay and I just purchased it.

Thank you.

onlize, 2017-07-07:
I confirmed that it was the relay. Thank you for your help.

closed 2017-07-07 by onlize