jerrod-lankford/GarageQTPi #13: Rebooting/Starting up Raspberry Pi triggers relay

ChadCurvin, 2017-09-05:
Hi!

Thanks for the great work! I've set this up on my Raspberry Pi Zero W and it works great, integrates perfectly with Home Assistant, everything works. 

However, when I reboot the Pi (or power is otherwise removed e.g. a power outage), then upon startup the script automatically "cycles" the pins and triggers the relay to open/close the garage door. Due to the stability of the Pi and this application, I only noticed this issue recently even though I have had this running for quite a while 👍 

Any suggestions? I'm ok with beginner Python, so my guess is it might have something to do with setting the pull_up_down to PUD_DOWN? Or would that just reverse the situation so that it will still cycle during startup? :)

It almost sounded like someone was having a similar issue but was in the wrong topic: 
https://github.com/Jerrkawz/GarageQTPi/issues/3
But I didn't see any resolution to their issue in there either.

Thanks!
-Chad

Jerrkawz, 2017-09-06:
Read the very last post in the thread and see if that is your issue. The home assistant configuration should not have the retain:true flag in it

Jerrkawz, 2017-09-06:
It's possible that there is another issue but check that first

ChadCurvin, 2017-09-10:
Thanks! 

This happens even without having it hooked up to Home Assistant, upon rebooting the Pi.

I did test by adding and changing "retain: True" and "retain: False" in my HA config.yaml with no effect (restarting Home Assistant after each change). Subsequent reboots of the Pi still caused the garage door to open.

Thanks for any further thoughts or things to try!
-Chad

Jerrkawz, 2017-09-11:
Hey,

So one thing I failed to mention, once a retained message is sent to a thread it is retained until its blown away. Can you change the command topic in the GarageQTPi configuration to point to something different that you haven't used before and then reboot the pi and see if it still opens?

If that doesn't work then its likely your wiring, or your pi. Im not sure familiar with the raspberry pi zero but I can look into it.

perJand, 2017-09-11:
Hi!
Thank you for a great work!

I have hade some issues with reconnection and reboot. Now I'm down to only when the garageRPi reboots. First I hade with reconnections also but it was solved with #3.

So, every time the garageRPi reboots it opens the doors (if they where closed).
I do not have any sensors, could that be the problem?

Jerrkawz, 2017-09-20:
The problem is likely that the pin you chose for the relay is on by default. Im not familiar with the pinouts on the pi zero, maybe you could find some documentation and try a different pin? If i have a chance i can also look

perJand, 2017-09-20:
I believe it is the main.py that triggers the pins on start up.
When I reboot without the autostart of main.py the do not go up (or down). Starting main.py manually triggers the pins.

Jerrkawz, 2017-09-25:
@perJand I just merged a new PR that allows support for inverted relay. Can you pull and try it? The default behavior is reversed now. Try it both ways in the config.yaml, probably with inveretd_relay: false first (since true used to be the default behavior)

Jerrkawz, 2017-09-25:
@ChadCurvin FYI ^^^

perJand, 2017-09-25:
Thank you, it works perfectly!
using invert_relay: false

Jerrkawz, 2017-09-25:
Glad to hear it! You can thank @tagno25 :)

closed 2017-09-25 by Jerrkawz

ChadCurvin, 2017-10-11:
Woohoo!! Just wanted to chime in here and mention that this fixed it for me as well. 

Thank you!