jerrod-lankford/GarageQTPi #14: CLOSE and OPEN explicitly?

ChadCurvin, 2017-10-11:
Hi!

I am wondering if I might have something setup incorrectly, or if it's just how this works currently. If I publish:

  `mosquitto_pub -h 192.168.1.100 -t home-assistant/cover/set -m "CLOSE"`

If the door is open, it closes. However, if the door is closed, it will open. GarageQTPi's main.py appears to correctly identify the garage door as being open or closed (based on the reed sensor / pins). So it seems like it's not respecting the command and is instead acting like:

  `mosquitto_pub -h 102.168.1.100 -t home-assistant/cover/set -m "TOGGLE"`

Do I have something misconfigured, or is this how it is currently setup?

Thank you!
-Chad

Jerrkawz, 2017-10-11:
Nope thats currently how its coded. I need to respect the current state of the pin i just haven't added that in.

Jerrkawz, 2017-10-11:
I'll leave this issue open and hopefully can address this soon when I get some free time.

closed 2017-10-12 by Jerrkawz