jerrod-lankford/GarageQTPi #16: Door Opening by Itself

v-rusmit, 2018-01-19:
Has anyone seen an issue with the Garage door opening by itself?  Its only happened a couple times but is concerning.  My guess is electrical interference could be triggering the GPIO but that's just a guess.

Jerrkawz, 2018-01-19:
Its probably not electrical. See threads like these https://github.com/Jerrkawz/GarageQTPi/issues/13

My guess would be that home assistant or whatever you have set up is publishing the OPEN/CLOSE command with a retain=true flag. So if the mqtt service restarts for whatever reason it will recieve the last command

v-rusmit, 2018-01-19:
Ok I'll turn off the retain flag.  It has happened so sporadically that I won't know its fixed until time has passed where it hasn't happened.  Thank you for your timely response!

closed 2018-01-19 by v-rusmit