jerrod-lankford/GarageQTPi #17: struct.error: cannot convert argument to integer

WalterWampe, 2018-02-14:
Hi there, 
I am quite frustrated atm and don't know what I am doing wrong...
This is what I get when I try python main.py:

> Welcome to GarageBerryPi!
> Traceback (most recent call last):
>   File "main.py", line 50, in <module>
>     client.connect(host, port, 60)
>   File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 768, in connect
>     return self.reconnect()
>   File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 940, in reconnect
>     return self._send_connect(self._keepalive, self._clean_session)
>   File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2125, in _send_connect
>     keepalive))
> struct.error: cannot convert argument to integer

Can you tell me what my problem is?
Thank you!!

WalterWampe, 2018-02-14:
> my config:
> 
> mqtt:
>     host: smartassi
>     port: 1883
>     user: garage
>     password: wampe
>     availability_topic: "garagepi/out/availability"
> doors:
>     -
>         id: 'garage'
>         relay: 17
>         state: 21
> #        state_mode: normally_closed #defaults to normally open, uncomment to switch
> #        invert_relay: true #defaults to false, uncomment to turn relay pin on by default
>         state_topic: "garagepi/out/state"
>         command_topic: "garagepi/in/set"
>         camera_topic: "garagepi/out/pictureStream"
>         image_File_path: "home/pi/motion/snapshot.jpg"

Jerrkawz, 2018-02-25:
Can you go to line 43 in main.py and change
port = CONFIG['mqtt']['port']
to
port = int(CONFIG['mqtt']['port'])

I dont know why this would be necessary but its the only thing i can think of. If that fixes it ill update main.py

closed 2018-04-28 by Jerrkawz

DanielNL, 2018-05-17:
Hey Jerrkawz,

I have the exact same errors when i try to start main.py. I can see you already closed the problem, so i hope the solution is found?

The only difference in my config.yaml is that by host i have the ip addres of my home assistant embeded mqtt broker set, might this be causing the error? 

Kind regards,

Daniel

Ps. I only just started to test out home assistant this week so i have a long way to go, be gentle :)

Jerrkawz, 2018-05-19:
I closed the issue because I never got a response from @WalterWampe. The lib.EventHook import seems wrong, it should just be EventHook since they are in the same dir. I can update. As for the paho mqtt issue I did post a potential solution above, i just have never tested it, If you want to you do what I suggested and see if that fixes the error.

DanielNL, 2018-05-19:
Hey Jerrod,

thanks for the response. At the start of the problem i did try your
suggestion to define it to an integer, no difference however.

Seeing as the pi i was installing this on was already running stable for 3
years the requirements file didn't work for me either, something to do with
http vs https. I installed both instances manually, albeit version 1.3.

When using the latest version (1.3) on a completly new rpi image i got the
same error again, so i switched back to 1.2 with seemed to fix that
specific error.

Now not being familiar with decent scripting i tried running the garage
file manually and the rest of my findings is in my latest eddit.

As of this afternoon i've got both doors running perfect within HA, wich
was my goal before i was willing to migrate from my old solution.

So all in all another experience had, thanks for taking the time to put
this all together. Higly appreciated!

Regards,

Daniel

Jerrkawz, 2018-05-19:
Glad you got it working. I commited the lib.eventHook to eventHook change and the int() change, event though it doesnt seem to help. Ill file an issue to figure out why it doesnt work with 1.3. Thanks!

nielsenaa, 2021-03-29:
Hi guys, The bug has something to do with setting (believe it or not) a particular protocol (MQTTv31 or MQTTv311) - check it out
i don't need a specific protocol version (and im not good enough in python right now) , but if it can help anyone else who would have the time to go through client.py for fixing the bug.. here you go, i narrowed it down to that by trial and error