jerrod-lankford/GarageQTPi #18: Parallel pushbutton didn't work anymore

geraldhansen, 2018-02-20:
Hi,

I installed your software on my working raspberrypi with https://github.com/andrewshilliday/garage-door-controller.
Turned the garage-door-controller of and start your version and configured them in home-assistant. Everything works fine and I can open and close my single garage door with home-assistant.

But - Parallel to the relay I have a push button installed who works fine in combination with garage-door-controller, but not with GarageQTPi. I just can trigger my push button once and afterwards he didn't react anymore. If I stop GarageQTPi everything is fine again and if I use garage-door-controller again also. Without changing the wiring.

Any Idea?
Thanks a lot.

tagno25, 2018-02-20:
Try enabling the "invert_relay" config line.

What may be happening it that the relay is closed acting like the button is constantly pressed.  And when you activate the garage door via home-assistant you are temporarily releasing the relay.

geraldhansen, 2018-02-21:
Yes, I tried already with "invert_relay" but also with the same behaviour. Everything works fine, except my push button.

Spartan-II-117, 2018-02-21:
Can you show your wiring diagram?

Jerrkawz, 2018-02-25:
Yes this does seem like a wiring issue. I wired my relay up to an extension cord and plugged a lamp into it to make sure that when i triggered the garage door that the lamp flipped on for a split second and then back off, its a good analogy to how the garage door works. Its hard to say whats going wrong without looking at your wiring.

geraldhansen, 2018-02-28:
Hi, this is how my wiring looks like.
[image]
And I don't think it's an wiring issue. Because it's working fine with the garage-door-controller. But I like to use mqtt. That's why I prefer your version.

Jerrkawz, 2018-03-06:
@geraldhansen Can you post your config that you used when setting invert_relay? garage-door-controller essentially is always in inverted relay mode so this is definitely what you want. If you have that flag set it should function exactly the same way.

Jerrkawz, 2018-03-12:
@geraldhansen Any update here?

geraldhansen, 2018-03-16:
Hi, sorry for the late answer. But it's currently very cold outside and in my garage ;-)
I did a rewiring and play around with invert config setup. I used the different pins on the relay and did some checks.
To leave this short - everything ist working now as expected. This isn't an issue for me anymore.
Thanks for your support and for your work.

closed 2018-03-16 by geraldhansen