jerrod-lankford/GarageQTPi #19: Problem with service

4ordy, 2018-03-16:
getting this error when the service is trying to run........

● garageqtpi@root.service - GarageQTPi
   Loaded: loaded (/etc/systemd/system/garageqtpi@root.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2018-03-16 07:05:06 UTC; 2s ago
  Process: 11297 ExecStart=/usr/bin/python /home/pi/GarageQTPi/main.py (code=exited, status=1/FAILURE)
 Main PID: 11297 (code=exited, status=1/FAILURE)

Mar 16 07:05:06 garagepi systemd[1]: garageqtpi@root.service: Failed with result 'exit-code'.

the bash script has run it has root....is that the problem or is there something in the code?

claughinghouse, 2018-03-16:
I am seeing the same issue on a Pi Zero W with Raspbian Stretch.

Spartan-II-117, 2018-03-16:
What command are you running that shows this output?

claughinghouse, 2018-03-16:
I was running `systemctl status garageqtpi@root.service`

Jerrkawz, 2018-03-16:
@tagno25 Was kind enough to write the service. Do you have any ideas?

claughinghouse, 2018-03-16:
I do not know much about python but I think there is an issue running `main.py` as root versus as the pi user. When I run `sudo python main.py` I get this error:

```root@GarageQTPi:/home/pi/GarageQTPi# sudo python main.py
Traceback (most recent call last):
  File "main.py", line 3, in <module>
    import yaml
ImportError: No module named yaml```

But when running without `sudo` this is the output:

```pi@GarageQTPi:~/GarageQTPi $ python main.py
Welcome to GarageBerryPi!
Connected with result code: Connection Accepted.
Listening for commands on home-assistant/cover/set```

claughinghouse, 2018-03-16:
I have resolved my issue. @4ordy I had to `sudo su` and then rerun `pip install -r requirements.txt` so that the requirements were installed as the `root` user that the `autostart_systemd.sh` installs the service as. 

Yours looks like the same as I had.

4ordy, 2018-03-17:
Works great, thanks so much @claughinghouse  and @Jerrkawz ...very happy!

closed 2018-03-17 by 4ordy