jerrod-lankford/GarageQTPi #21: Systemd service file should start python with -u parameter

reibuehl, 2018-04-28:
By default, python seems to buffer output which leads to systemd/journald not receiving the results of the prints to STDOUT in a timely manner. If the service file provided would start the script with python in unbuffered mode
`ExecStart=/usr/bin/python -u /home/pi/GarageQTPi/main.py `
the logging to journald would come immediately and not in larger junks sometime or never.

Jerrkawz, 2018-04-28:
Want to create a pr?

reibuehl, 2018-04-28:
I was already doing it but then I thought I should also fix the pin numbering in the example (done) and on the wiring diagram but then I somehow got lost finding the original and its license... :-)
I was tempted to just change the numbers in the picture but don't want to violate anybody's copyright. Maybe I recreate it and add it as a new file in the PR if you don't mind?

closed 2018-05-19 by Jerrkawz