jerrod-lankford/GarageQTPi #26: No connection through GPIO pins

smeranda, 2019-01-07:
I have this installed on Raspberry Pi Zero, no issues with the installation.

Regardless of my magnetic switch (I have the same ones you documented), it's not registering in GarageQTPi correctly. In other words, if I have the switches in contact, it still thinks the garage is open. If I switch the `state_mode`, the state never changes regardless of the state of magnetic switches.

Furthermore, the commands to open and close are registering (as evidenced by the logs), but the signal never makes it to the garage door.

I don't think this is a software issue (at least not with GarageQTPi), it feels more like a GPIO issue. However, this is a brand new board with the headers soldered on professionally at the factory. 

Do you have any suggestions on how to troubleshoot connectivity issues?

Jerrkawz, 2019-01-07:
Hi, I would maybe try http://wiringpi.com/the-gpio-utility/ and see if you can manually read/write to the gpio pins

smeranda, 2019-01-07:
Thanks for the reply! I used WiringPi to run `pintest`:

```
          The main 8 GPIO pins  0: 7:  OK
                The 5 SPI pins 10:14:  OK
               The serial pins 15:16:  OK
                  The I2C pins  8: 9:  OK
```

It seems the pins are functioning? Suggestions on where else to look?

reibuehl, 2019-01-21:
Do the values in the "V" column change, when you open/close the magnetic switches?

WWE-Corey, 2019-10-03:
Did you figure this out? I am thinking of building using a Pi Zero W as well.

smeranda, 2019-10-03:
Nope, I never did. I gave up and moved onto other projects.