jerrod-lankford/GarageQTPi #27: ValueError: Channel must be an integer

JBenzie, 2019-01-26:
I have GarageQTPi installed on a Raspberry Pi B running Raspbian Stretch Lite. I'm also running a Raspberry Pi 3 with Hass.io installed. I'm wanting to control my garage door from within Home Assistant using GarageQTPi over MQTT.

When I run `python main.py` from within the GarageQTPi directory, I get the following Traceback:
`Traceback (most recent call last):
    File "main.py", line 61, in <module>
       door = GarageDoor(doorCfg)
    File "/home/pi/GarageQTPi/lib/garage.py", line 31, in __init__
       GPIO.setup(self.relay_pin, GPIO.OUT)
ValueError: Channel must be an integer or list/tuple of integers`

My config.yaml file looks like this:

`mqtt:
        host: 192.168.1.136
        port: 1883
        user: homeassistant
        password: my_password
doors:
        -
             id: 'garage_door'
             relay: '23'
             state '17'
             state_topic: "home-assistant/cover/garage_door"
             command_topic: "home-assistant/cover/garage_door/set"
             position_topic: "home-assistant/cover/garage_door/set_position"
             availability_topic: "home-assistant/cover/garage_door/availability"
`

I'm new to using MQTT, so please let me know if anything is incorrect/incompatible with my config. I've added the position_topic and availability_topic only since they were included in the Home Assistant config and I figured everything should be identical between the two. 

Any ideas as to why I'm getting the error? I've been unsuccessful getting anything other than "unavailable" or "unknown" as a status for the garage door in Home Assistant.

Thanks in advance,
jerry

tagno25, 2019-01-26:
The relay and and state values should be just the numbers without apostrophes or quotes, just like the port is.

JBenzie, 2019-01-26:
Thank you. That solved that issue. I'm still not getting an updated status in Home Assistant, though. I'm sure that's another issue.

Thanks for the help.

closed 2019-04-14 by Jerrkawz