jerrod-lankford/GarageQTPi #28: Incorrect protocol name

Sleeepy2, 2019-02-07:
I hope this is still being watched as I just found it ;)

Installed and setup on the pi without an issue (I think) but I am getting this error in HA.

Invalid connection from (client @=192.168.1.137:37667) : [MQTT-3.1.2-1] Incorrect protocol name: "MQIsdp"
1:56 PM util/async_.py (ERROR)

Invalid connection from (client @=192.168.1.137:60523) : [MQTT-3.1.2-1] Incorrect protocol name: "MQIsdp"
1:56 PM util/async_.py (ERROR)

This error continuously happens with different ports.

Currently running HA version 0.86.4 and using the embedded MQTT.

tagno25, 2019-02-07:
In main.py change protocol=3 to protocol=4 (on line 45).

Sleeepy2, 2019-02-07:
Wow, thanks for the quick response. That has fixed this issue.
Thanks

closed 2019-02-07 by Sleeepy2