jerrod-lankford/GarageQTPi #29: Unable to run main.py

8osman, 2019-03-19:
I've installed everything onto my pi.
But when running main.py I get

```
Welcome to GarageBerryPi!
Traceback (most recent call last):
  File "main.py", line 45, in <module>
    user = CONFIG['mqtt']['user']
TypeError: string indices must be integers, not str
```
I seemed to be getting lots of errors with the config.yaml but have been able to resolve those just by changing the spaces..... however this error could be related. I'm installing via ssh from a windows 10 computer.

8osman, 2019-03-19:
Just re-reading the instructions and it says a pi 3 to be used. Think the one I have out there is a pi 2b is that going to make a difference?

8osman, 2019-03-20:
I've re-downloaded everything, re-installed everything. I've stripped back the config.yaml to only the mqtt logon details.

I'm still getting the same error. I'm confident now this is a programming issue..... it's just outside of my knowledge base.  :-(

8osman, 2019-03-20:
okay fixed the first fault
```
mqtt:
    host: 192.168.0.3
    port: 1883
    user: homeassistant
    password:
```
I hadn't got a space between the 'user:' and the user name. 

however I'm onto problem 2 now.

```
Welcome to GarageBerryPi!
Traceback (most recent call last):
  File "main.py", line 70, in <module>
    if not doorCfg['name']:
TypeError: string indices must be integers, not str
```
Spacing all looks good this time!? Maybe

my doors name id etc from config.py

```
doors:
    id: 'left'
    name: 'left Garage Door' #defaults to an unsanitized version of the id paramater
    relay: 23
    state: 17
    state_mode: normally_open #defaults to normally open, uncomment to switch
    invert_relay: false #defaults to false, uncomment to turn relay pin on by default
    state_topic: "home-assistant/cover"
    command_topic: "home-assistant/cover/set"
    id: 'right'
    name: 'right Garage Door' #defaults to an unsanitized version of the id paramater
    relay: 24 
    state: 27 
    state_mode: normally_open #defaults to normally open, uncomment to switch
    invert_relay: false #defaults to false, uncomment to turn relay pin on by default
    state_topic: "homeassistant/cover"
    command_topic: "homeassistant/cover/set"

```

This is being a little tricky...

8osman, 2019-03-28:
Okay, the culprit was spacing issues, re-copied the config.yaml and filled in the details and it's now working.

So initial loading problems are all sorted now. 

My garage doors are wired up ref Andrew Shillidays "my garage door controller" which I've been using with no problems for the last 2 years.

Running this opens the garage doors on program load I altered the state mode and that seems to have stopped that behaviour, unfortunately my manual push buttons in the garage no longer work while this program is running. 

I've not been able to work out where the differences are between the "my garage door controller" program and the "GarageQTPi" program, they seem similar. However no problems with the former. 

I'm going to try and  see if I can program home assistant via the http params directly into "my garage door controller" as this seems to work better for me. 

All detected fine by home assistant, using mqtt though.

tagno25, 2019-03-28:
If the garage door is opening on load, and the button stops working, then you probably need to set "invert_relay: true"

Jerrkawz, 2019-04-14:
Yes tagno is right. This boolean is likely the fix

closed 2019-04-14 by Jerrkawz