jerrod-lankford/GarageQTPi #31: Can't find installed libraries when running as a service

bg1000, 2019-05-31:
I wired things up in my garage and then ran a couple of test scripts to verify the wiring and GPIO numbers were as expected.

I followed the installation instructions in README.md.
Running the app from the command line (python main.py) works exactly as expected with no errors.  I enabled discovery and was also able to add the garage cover to the HA front end. I can open and close the door and the icon changes as expected.

I ran `sudo bash autostart_systemd.sh` which ran without error.  On reboot I see the following in syslog:

`May 30 19:45:52 raspberrypi python[415]: Traceback (most recent call last):
May 30 19:45:52 raspberrypi python[415]:   File "/home/pi/GarageQTPi/main.py", line 3, in <module>
May 30 19:45:52 raspberrypi python[415]:     import yaml
May 30 19:45:52 raspberrypi python[415]: ImportError: No module named yaml
May 30 19:45:52 raspberrypi systemd[1]: garageqtpi@root.service: Main process exited, code=exited, status=1/FAILURE
May 30 19:45:52 raspberrypi systemd[1]: garageqtpi@root.service: Unit entered failed state.
May 30 19:45:52 raspberrypi systemd[1]: garageqtpi@root.service: Failed with result 'exit-code'.`

This repeats every 3 seconds when the service is restarted.  It seems that when running as a service that python can not find the libraries that were installed via pip install -r requirements.txt.  I gather this has something to do with how sys.path is set but didn't completely work through that.

After some research I found this post on stack exchange which describes the exact same issue (different library):
https://stackoverflow.com/questions/38527235/running-python-script-via-systemd-fails-to-load-module

The answer in post 8 is to install the libraries via sudo -H pip install -r requirements.txt
Things work as expected now. I am not 100% certain if this is something specific to how my environment is setup or if others might run into the same issue.

Coch, 2019-06-16:
Just had the same issue when trying to get this setup in my environment - installing the libs under root via 'sudo su - root' or using the '-H' as outlined above seemed to do the trick....

tagno25, 2019-06-16:
Could you test if "sudo echo ${SUDO_USER:-${USER}}" reports the correct username.

bg1000, 2019-06-17:
Not sure which one of us you were asking. In my case when I log in as pi "sudo echo ${SUDO_USER:-${USER}}" returns "pi".

Coch, 2019-06-17:
As it does for me -- that said...

`#!/bin/bash
cp garageqtpi@pi.service /etc/systemd/system/garageqtpi@`whoami`.service
sed -i "s?/home/pi/GarageQTPi?`pwd`?" /etc/systemd/system/garageqtpi@`whoami`.service
systemctl --system daemon-reload
systemctl enable garageqtpi@`whoami`.service`

All of those examples result in returning root...not sure it's necessarily relevant as it looks like more of a naming thing than anything else...

tagno25, 2019-06-17:
#32 should fix the issue, please try the changes after removing the old/incorrect service using the commands below, and report back.

```
systemctl disable garageqtpi@root.service
rm /etc/systemd/system/garageqtpi@root.service
```

The following is the updated script

```
#!/bin/bash
cp garageqtpi@pi.service /etc/systemd/system/garageqtpi@${SUDO_USER:-${USER}}.service
sed -i "s?/home/pi/GarageQTPi?`pwd`?" /etc/systemd/system/garageqtpi@${SUDO_USER:-${USER}}.service
systemctl --system daemon-reload
systemctl enable garageqtpi@${SUDO_USER:-${USER}}.service
```

bg1000, 2019-06-18:
Thanks for your work on this.  I will give this a try and report back.  It probably won't be until this weekend. The RPI I have this installed on was kind of a dev box for a while and has a lot going on.  I am in the process of building a clean install for it offline.  If everything goes well I will try the updated script on the clean install as well (will likely be a better test).

bg1000, 2019-07-06:
I wanted to do a little belated follow up in this. Today I finally got around to rebuilding this system from scratch.  I started with a clean Raspbian image did a sudo apt-get update && sudo apt-get upgrade and then installed the app using the directions in the readme.  I coped my old config file over, started the service using systemctl and it worked perfectly on the first try. Thanks again for the follow up on this. I am closing this issue.

closed 2019-07-06 by bg1000