jerrod-lankford/GarageQTPi #37: Can't actually trigger the relay

andystewart999, 2020-02-02:
Hi - I've got a weird issue here.

Everything is set up fine, with discovery turned off.  HA shows the Cover and if I open and close the garage door the 'state change' event fires and HA shows the appropriate state - so the MQTT comms are working OK.

However, if I click on any of the up/down/stop buttons on HA, nothing appears to be getting sent to the other Pi!  For testing I'm running main.py interactively so I'd expect at least to see the 'connection' and 'listening' messages but they aren't coming up either.

Any thoughts about what's going on here?

Cheers,
Andy

andystewart999, 2020-02-04:
An update - I did some testing with discovery and verbose logging in HA.

Discovery is working fine and HA is still receiving status change messages.  I've turned on debug logging for MQTT and can see the OPEN/CLOSE message going out but each Transmit is followed straight away by a Receive:

_> 2020-02-04 22:39:29 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on homeassistant/cover/Garage/set: CLOSE
> 2020-02-04 22:39:29 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/cover/Garage/set: b'CLOSE'

I thought that was a bit odd so I killed main.py on the Pi, tried again and I STILL get the same messages!  It appears that HA is sending the messages to itself?!

bg1000, 2020-02-04:
@andystewart999 
I'm not the original author but have used this for some time and can make a couple of suggestions for you.  

1) If you have already run autostart_systemd.sh then garageqtpi is being started as a service.  This means if you are also running it interactively it's running twice.  If you want to run it interactively I recommend temporarily stopping the service with ```sudo systemctl stop garagqtpi@pi```. You can disregard this if you haven't yet run autostart_systemd.sh.
2) If you look at main.py you will see this callback:
```def on_connect(client, userdata, rc):
    print "Connected with result code: %s" % mqtt.connack_string(rc)
    for config in CONFIG['doors']:
        command_topic = config['command_topic']
        print "Listening for commands on %s" % command_topic
        client.subscribe(command_topic) 
```
Are you seeing the Connected. . . message?  If not then garageqtpi is not connecting to the broker and getting a response back.  You should also see a listening . . .  message for each door. If you're not connecting to the broker or you don't see at least one listening message. I would probably start by looking at the configuration in the yaml file (garageqtpi not HASS).
3) If you are connecting to the broker I would suggest looking at the mqtt topics directly rather than only looking at the home assistant log (which is useful but perhaps not the whole story).  You can do this by going to developer tools in the front end and clicking mqtt.  There you can subscribe to topics and see the actual message flow.  You might find it useful to use a wildcard like ```homeassistant/cover/#``` You can also send messages to topics and watch for a response. This allows you to simulate what HASS should be doing and check for a garageqtpi response as well as the other way around.

andystewart999, 2020-02-04:
Hi @bg1000 , thanks for the info.

1 - yep, haven't got the service running yet, for the initial testing I'm just running it locally so I can see the debug/console message.

2 - yep, I've gone through the code and noted all the connect/subscribe messages... and they AREN'T firing.  Obviously either the callback isn't registering or the client isn't connecting.  But I know it IS connected as events from the client (garage door opening) ARE being noted and registered on the HA server via MQTT.  It's just one-way communication right now for some reason (client to HA)

3 - good suggestion, I think I've already tried that but I'll have another crack via the dev tools as soon as I can.

So odd - obviously there's a connection but it's not 100% bi-directional!

Final-Hawk, 2021-01-31:
Same issue, found a solution?