jerrod-lankford/GarageQTPi #39: is RPi3 really a min requirement?

vulcanjedi, 2020-06-06:
Saw that  a PiZeroW seems like it should work?
I actually have an older RPi Model B rev 2 that I'd love to repurpose and get use for if it would work?

bg1000, 2020-06-07:
I suspect that would probably be fine. The app is doing minimal processing and data transfer.  I am running it on a 3B that I already owned. It is running several other things and still isn't very heavily loaded. You will have to test it to be sure of course. 

The original author has told me he doesn't use this anymore. My fork is more up to date and includes support for python3 and some additional features so feel free to check that out.

Neo-Rules, 2020-06-08:
I used a PzeroW on mine last year. Works great

vulcanjedi, 2020-06-08:
Thanks friends. Got into HA a while back and always want to do some more w/ my RPis.  I am using the others but was hoping to repurpose this og RPI.
Finally got my wire/relay
I got dupont connectors.  Trying to test before going to the trouble to mount /solder everything. But not getting anythign on mqtt?
The Cover on MQTT shows up but I only have the Pi and on switch wired up and the input of the relay wired up.
Was also wondering if I could wire an additional reed switch for the inside door? 
Is that GPIO 22 w/ 25?

reibuehl, 2020-06-11:
I run it on a Raspberry Pi A, so a Raspberry Pi 3 is not required. Any model should do.

vulcanjedi, 2020-06-11:
Thanks gents
I think I figured out the code part. 
Couldn’t get the fork to work so has to stick with original
One of the openers is huge pita so gonna try from the wall mount button approach

vulcanjedi, 2020-06-13:
Definitely works on the RPI Model B rev 2

closed 2020-06-13 by vulcanjedi