jerrod-lankford/GarageQTPi #43: Any chance for help running this prog ?

rapi3, 2020-11-14:
Hi, any chance to get help to have this running ?
Python 2.7.9
Requirement already satisfied: paho-mqtt in /usr/local/lib/python2.7/dist-packages (1.5.1)
Requirement already satisfied: PyYAML in /usr/local/lib/python2.7/dist-packages (5.3.1)
```
python main.py
Welcome to GarageBerryPi!
main.py:42: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  CONFIG = yaml.load(ymlfile)
Traceback (most recent call last):
  File "main.py", line 108, in <module>
    client.loop_forever()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1779, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1181, in loop
    rc = self.loop_read(max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1572, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2310, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2942, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 3030, in _handle_connack
    self, self._userdata, flags_dict, result)
TypeError: on_connect() takes exactly 3 arguments (4 given)

rapi3, 2020-11-14:
Maybe it help to debug further I can see this on HA MQTT log:
```
1605372533: New client connected from 192.168.66.3 as MQTTGarageDoor_13453437c7b5 (p1, c1, k60, u'rpimqtt').
1605372533: Socket error on client MQTTGarageDoor_13453437c7b5, disconnecting.

bg1000, 2020-11-14:
@rapi3 I am running this fork https://github.com/bg1000/GarageQTPi but can try to help.  Can you publish the content of your config file?

rapi3, 2020-11-14:
HI,
thank you very much for helping me.
I test now your fork and it work for first try ! Good Job !

My goal is to have a program like this running on rpi that will allow me to integrate in HomeAssistant so I can create basic automation.
My remote rpi have few relays connected to RF remote plug that at this moment I command ( ON sleep 1s OFF ) with simple bash script or python from crontab or from a changed version of GarageController.
So I can't have any feedback with other relays.

bg1000, 2020-11-14:
Glad it worked for you.  Please feel free to open an issue on the fork if you have any problems.

rapi3, 2020-11-14:
Do you have any idea if there is available an sw like Tasmota that run on rpi ( python... ) where we can define more relays, sensors, and how to function; switch,cover, sensor... and will integrate with Home Assistant ?
Thank you.

rapi3, 2020-11-15:
I think I found what I am looking for my project to work:
```
https://github.com/flyte/pi-mqtt-gpio

closed 2020-11-15 by rapi3