jerrod-lankford/GarageQTPi #53: main.py is full of issues due to newer versions of PiOS

Danielson89, 2024-09-16:
File "main.py", line 41
    with open(os.path.join(os.path.abspath(os.path.dirname(__file__)), 'config.yaml'), 'r') as ymlfile:
    ^
IndentationError: expected an indented block

I've also commented out all the print command lines as they don't work

Danielson89, 2024-09-19:
Had to remove this got sick to trying to resolve every single issue that kept coming up,  fix one thing another issue,  almost ended up with a blank .py script at the end which had no issues

vulcanjedi, 2024-09-19:
I went with Shellly 1 plus. So much less headache after my original pi died. Tried to replace with nodemcus but was such a pita.  Went with Shelly out of the box. Kept all the existing wiring. I had. No coding or updates or hacks. One confitg change in Shelly gui. More expensive than nodemcus or maybe even pi but so essy